someweisguy/esp_dmx #162: esp32c6 Not working

Pavan-keus, 2024-07-29:
Esp32c6 on build get some error like dmxTest/dmxTest/components/esp_dmx/src/dmx/hal/uart.c:346:5: error: 'LP_CLKRST' undeclared (first use in this function)

riwalker, 2024-07-31:
Ensure you have the latest esp_dmx vs 4.1 with the fix:

https://github.com/someweisguy/esp_dmx/pull/158

 

and also the latest ESP-IDF > = 5.3

 

This was a uart.c change for the C6 with the 3rd UART being a LP UART

Ensure you upgrade to the latest esp-idf, mine is 5.3

There have been a lot of changes in the recent esp-idf from 5.1 onwards for the support for the LP UARTS, hence all the changes

 

This is the code that is flagging the issue:

 

#if CONFIG_IDF_TARGET_ESP32C6

  // UART2 on C6 is a LP UART, with fixed GPIO pins for tx, rx, and rts

  if (dmx_num == 2) {

    LP_CLKRST.lpperi.lp_uart_clk_sel = 0;  // Use LP_UART_SCLK_LP_FAST

  } else {

    uart_ll_set_sclk(uart->dev, UART_SCLK_DEFAULT);

  }

  uart_get_sclk_freq(UART_SCLK_DEFAULT, &sclk_freq);

#else

Pavan-keus, 2024-08-05:
yeah , it's working fine for version 5.2.2 but i make sure to make it 5.3 
thnx for reply @riwalker

closed 2024-08-05 by Pavan-keus