someweisguy/esp_dmx #167: DMXWrite Example, Byte doesn't store 512 values + Suggestions

jhsa, 2024-09-15:
Hi, Just noticed that on theDMXWrite example "data" is declared as byte, which hold only 256 values. DMX channels have 512 channels. See below:

`#include <Arduino.h>
#include <esp_dmx.h>

int transmitPin = 17;
int receivePin = 16;
int enablePin = 21;

/* Next, lets decide which DMX port to use. The ESP32 has either 2 or 3 ports.
  Port 0 is typically used to transmit serial data back to your Serial Monitor,
  so we shouldn't use that port. Lets use port 1! */
dmx_port_t dmxPort = 1;

/* Now we want somewhere to store our DMX data. Since a single packet of DMX
  data can be up to 513 bytes long, we want our array to be at least that long.
  This library knows that the max DMX packet size is 513, so we can fill in the
  array size with `DMX_PACKET_SIZE`. */

**byte data[DMX_PACKET_SIZE]; // byte can only store 256 values from 0 to 255, not 512 values**

/* This variable will allow us to update our packet and print to the Serial
  Monitor at a regular interval. */
unsigned long lastUpdate = millis();

void setup() {
  /* Start the serial connection back to the computer so that we can log
    messages to the Serial Monitor. Lets set the baud rate to 115200. */
  Serial.begin(115200);

  /* Now we will install the DMX driver! We'll tell it which DMX port to use,
    what device configuration to use, and what DMX personalities it should have.
    If you aren't sure which configuration to use, you can use the macros
    `DMX_CONFIG_DEFAULT` to set the configuration to its default settings.
    Because the device is being setup as a DMX controller, this device won't use
    any DMX personalities. */
  dmx_config_t config = DMX_CONFIG_DEFAULT;
  dmx_personality_t personalities[] = {};
  int personality_count = 0;
  dmx_driver_install(dmxPort, &config, personalities, personality_count);

  /* Now set the DMX hardware pins to the pins that we want to use and setup
    will be complete! */
  dmx_set_pin(dmxPort, transmitPin, receivePin, enablePin);
}

void loop() {
  /* Get the current time since boot in milliseconds so that we can find out how
    long it has been since we last updated data and printed to the Serial
    Monitor. */
  unsigned long now = millis();

  if (now - lastUpdate >= 1000) {
    /* Increment every byte in our packet. Notice we don't increment the zeroeth
      byte, since that is our DMX start code. Then we must write our changes to
      the DMX packet. */
    for (int i = 1; i < DMX_PACKET_SIZE; i++) {
      data[i]++;
    }
    dmx_write(dmxPort, data, DMX_PACKET_SIZE);

    /* Log our changes to the Serial Monitor. */
    Serial.printf("Sending DMX 0x%02X\n", data[1]);
    lastUpdate = now;
  }

  /* Now we can transmit the DMX packet! */
  dmx_send_num(dmxPort, DMX_PACKET_SIZE);

  /* We can do some other work here if we want. */

  /* If we have no more work to do, we will wait until we are done sending our
    DMX packet. */
  dmx_wait_sent(dmxPort, DMX_TIMEOUT_TICK);
}
`

Also, if someone is a beginner like me, it would help to understand the code better if more used DMX names were using. For example "Channel" instead of "data", "NrOfChannels" instead of "DMX_PACKET_SIZE"
Please do not take it as criticism, these are just suggestions to make the library easier to understand.
Thanks for the library..

tommyco10, 2024-09-17:
An individual dmx data slot is 0 indexed byte,  representing a value between 0 and 255. A DMX512 frame has 513 bytes, with slot[0] representing special startcode data and slots data[1] to data[512] representing 8 bit values for addressable parameters of a DMX512 receiving device or devices.

Essentially a DMX frame is an array of 513 individual bytes transmitted sequentially, so in this context:
```
byte data [DMX_PACKET_SIZE]
```
 is an array of bytes, with the square brackets denoting how many of them there are according to the value of DMX_PACKET_SIZE.

jhsa, 2024-09-28:
Thank you very much for your reply, and for the explanation. Now I understand that there is a "Start Byte" that actually dictates the type of data that will follow..

tommyco10, 2024-09-29:
No problem at all, happy to help. 

As you mentioned the start code determining how the data which follows should be interpreted (0x00 for DMX data or 0xCC for RDM) it's worth noting that manufacturers can register for their own Alternative Start Code to transmit manufacturer specific data, which can be found [here](https://tsp.esta.org/tsp/working_groups/CP/DMXAlternateCodes.php)