someweisguy/esp_dmx #173: rdm_sensor_definition_add fails in sensor example

ebeam-bob, 2024-10-11:
Love what you've done with v4.1.0, and there's so much there now that I'm still digesting it all!
I built the Arduino_RDMSensor example, and from Device Info on my Botex RDM tool I see that there is one sensor (the Sub-Dev count is 0) but it does not give me the option to read the sensor.
Then I put the call to rdm_sensor_definition_add into an if statement with a println, and sure enough that call is returning false. 
Is something awry in the example, or in that function?

ebeam-bob, 2024-10-18:
The error is coming from the test "sensor_defs[definition->num].num != 0xff", but if I comment that out, the ESP32 keeps resetting with these messages:
   Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
   Debug exception reason: Stack canary watchpoint triggered (loopTask)

ebeam-bob, 2024-10-28:
In rdm_register_sensor_definition, the first_time_func_called flag is false the first time this is called and so the loop that sets the sensor definition number to unused (sensor_defs[i].num = 0xff;) did not run.

ebeam-bob, 2024-11-06:
In rdm_rhd_get_sensor_definition, the second argument to rdm_read_pd should be definition->get.request.format, not set.request.format, which is NULL.

riwalker, 2025-10-29:
Has anyone got an example of sensors working ?  code builds, and KISS (Keep it simple), using basic [RDMControl](http://www.hoelscher-hi.de/hendrik/english/openrdm.htm),  the sensors are gibberish, 
no sensor name, and the values are wrong
(I know the tool works as have other RDM code library on another micro publishing fine)

[image]

riwalker, 2025-10-30:
see fixes in #175