someweisguy/esp_dmx #177: Timer.h / Responder: Response -> Controller: Any Packet

kudiltz, 2024-10-25:
Testing RDM controller against a number of responders hung the UART on two different responders.  Most have not had this issue, though.

Looking closer, the timing between response to mute and the next controller transaction appears to be 77uS:
[image]

From the standard at least 176 is required:
[image]

I see nearly all of the other timing requirements in this table are accounted for.  Am I missing this one?

Generally very good marks on the rest of the tests I've run the controller through! So, thank again.

ebeam-bob, 2024-10-28:
@kudiltz What tool did you use to trace all that packet timing?  I'd like to know what hardware and software you used.

kudiltz, 2024-10-31:
> @kudiltz What tool did you use to trace all that packet timing? I'd like to know what hardware and software you used.

@ebeam-bob I had an opportunity to use a DMXter4A RDM from Goddard: http://www.goddarddesign.com/
which has USB connection to teminal emulator like Terra Term.