someweisguy/esp_dmx #178: Need a hint on DMX packet transmission update rate using Artnet + esp_dmx libraries

knopserl, 2024-11-01:
Hi, sorry for this basic question but do I understand it right, that the values of all DMX channels need to be constantly transmitted in a certain frequency like 25 to 40Hz and not just if any value of the channels has been changed?
I'm using two libraries: one for Artnet and this one for sending the received Artnet values to one DMX universe.
Artnet library used is this one: https://github.com/hideakitai/ArtNet (it supports Ethernet which I use and WiFi and also supports poll/response, discovery)
In the loop, I also call artnet.parse which calls a callback, where I take the values from artnet and wrting/sending it to the DMX bus.
I don't resend the values in the loop. So I guess I don't to an cyclic update on the DMX bus if no value has changed.
Is that DMX compliant or do I need to repeat sending out the current values in a frequency of 40Hz or does the esp_dmx library do it?

Attached the source of the main sketch, the artnet and the DM part.
I hope you understand my question and can provide me with a proper advise.
BTW: I noticed that my Chineese Test DMX moving head device frequency goes back to a inital state and then gets the current setting agin after a few seconds (I think the artnet interval does that refresh and resend the values). This was the trigger for that question
   
[ESP32-ETH-Artnet.zip](https://github.com/user-attachments/files/17603141/ESP32-ETH-Artnet.zip)

k-dedman, 2024-11-07:
It depends on how you are connected to your DMX fixtures. If you are directly connecting the UART output to a RS485 transiever, you will need to call dmx_send or dmx_wait_sent in a loop. If you are using a 3rd party converter specifically for DMX, it may hold the last packet in memory and repeat it if it doesnt receive a new one in the required time. DMX fixtures require continuous packets or else they will turn off, the fixture im testing with turns off after ~1s of no input.

knopserl, 2024-11-07:
> It depends on how you are connected to your DMX fixtures. If you are directly connecting the UART output to a RS485 transiever, you will need to call dmx_send or dmx_wait_sent in a loop. If you are [...]

Thanks, that was the answer I was expecting.
I directly connect my ESP32 sketch using esp_dmx via UART + RS485 (MAX485) to the fixtures and I have obsered, that they turn to the default settings if no new DMX data are send by my sketch because no changes where made.
So I need to resend the current (all) universe values constanly in the loop or are least with 25-45 FPS/Hz.