someweisguy/esp_dmx #181: doesn't compile on esp-idf 5.3

juanpgg-git, 2024-11-20:
`C:/Users/juanp/OneDrive/Documentos/Digital-dots/firmware-lights/components/esp_dmx/src/dmx/hal/uart.c: In function 'dmx_uart_init':
C:/Users/juanp/OneDrive/Documentos/Digital-dots/firmware-lights/components/esp_dmx/src/dmx/hal/uart.c:330:51: error: 'uart_signal_conn_t' has no member named 'module'
  330 |   periph_module_enable(uart_periph_signal[dmx_num].module);
      |                                                   ^
C:/Users/juanp/OneDrive/Documentos/Digital-dots/firmware-lights/components/esp_dmx/src/dmx/hal/uart.c:338:52: error: 'uart_signal_conn_t' has no member named 'module'
  338 |     periph_module_reset(uart_periph_signal[dmx_num].module);
      |                                                    ^
C:/Users/juanp/OneDrive/Documentos/Digital-dots/firmware-lights/components/esp_dmx/src/dmx/hal/uart.c: In function 'dmx_uart_deinit':
C:/Users/juanp/OneDrive/Documentos/Digital-dots/firmware-lights/components/esp_dmx/src/dmx/hal/uart.c:384:56: error: 'uart_signal_conn_t' has no member named 'module'
  384 |     periph_module_disable(uart_periph_signal[uart->num].module);`

tresler, 2024-11-28:
Yes, in IDF 5.3 structure uart_signal_conn_t now don't have member "module". Have only IRQ and pins:
[IDF 5.3](https://github.com/espressif/esp-idf/blob/release/v5.3/components/soc/include/soc/uart_periph.h) vs [IDF 5.1](https://github.com/espressif/esp-idf/blob/release/v5.1/components/soc/include/soc/uart_periph.h)

b-destoop, 2024-12-07:
It looks like this is a bug on the ESP-IDF side. 

`esp_dmx/src/dmx/hal/uart.c` refers to `uart_periph.h` and expects to find a definition of `uart_signal_conn_t` that contains a 'module' member. Except it does not exist in the version of `uart_periph.h` as given.

`uart_periph.h` is both found in 
`/esp-idf/components/soc/include/soc/uart_periph.h` 
and in
`/home/bert/esp/esp-idf/components/soc/esp32/uart_periph.c`
The second version contains the right definition of `uart_signal_conn_t` as expected, but is included for the compiler AFTER the first version, so the wrong content is selected.

Up until commit dca7c286d009024d8bacab52425c93a91ffc49c7, the version that was in /components/soc/include/soc/uart_periph.h did contain the expected definition, but was removed. [Commit found here](https://github.com/espressif/esp-idf/commit/dca7c286d009024d8bacab52425c93a91ffc49c7)

tresler, 2025-01-28:
I found this [fork](https://github.com/davispolito/esp_dmx/tree/release/v4.1) and use #define macro for IDF version and everything work fine on ESP32-c6 with arduino-esp32 version 3.1.1. Only change .txt to .c

[uart.txt](https://github.com/user-attachments/files/18575376/uart.txt)

gunstr, 2025-02-06:
Just realized that @davispolito already has subitted a PR (#187) with his changes...