someweisguy/esp_dmx #184: UART1 interrupt is not deallocating

SolidStateLEDLighting, 2024-12-11:
I am going through a repeated process where I establish and tear down a DMX service.   And I'm doing something wrong because I'm getting: **intr_alloc:  No free interrupt inputs for UART1 interrupt (flags 0x40E)**    every 3 cycles.

To establish I used this sequence:
1) dmx_driver_install
2) dmx_set_pin

To tear down I use this sequence:
1) dmx_driver_disable
2) dmx_driver_delete

How can I make sure all resources are freed?

SolidStateLEDLighting, 2024-12-13:
I am running multiple core (s3) on the IDF v5.3.1.

In addition to this deallocation problem, I also can't send more than one frame of data.   I suspect that dmx.status never returns to DMX_STATUS_IDLE.

bensuffolk, 2025-02-11:
Pull request https://github.com/someweisguy/esp_dmx/pull/191 fixes this issue.

SolidStateLEDLighting, 2025-02-12:
Which issue?    The inability to reclaim resources properly or the inability to send more then one frame of data?    OR - are both issues solved?

bensuffolk, 2025-02-12:
The main issue you reported initially (dealloc of resources). I didn’t actually see the in addition bit.

I had already come across the resources thing and fixed it in my project so didn’t really focus on your entire message, sorry. Although I don’t have that problem and am running on an s3 with 5.3.0

Will try a later version.

Ben