someweisguy/esp_dmx #186: Arduino_DMXWirte example compilation error

Jibeji, 2025-01-25:
Hi,
Trying to get the example working on an ESP32 Dev Module board with Arduino IDE 1.8.9 and the latest Espressif definitions, I have this error:

`C:\Users\JB\Documents\Arduino\libraries\esp_dmx\src\dmx\hal\uart.c: In function 'dmx_uart_init':
C:\Users\JB\Documents\Arduino\libraries\esp_dmx\src\dmx\hal\uart.c:330:51: error: 'uart_signal_conn_t' has no member named 'module'
  330 |   periph_module_enable(uart_periph_signal[dmx_num].module);
`

tracetechnical, 2025-01-25:
Same issue here. It seems going to esp32 3.0.4 in boards manager and 4.0.0 of this library makes the error disappear. But I haven't checked that it actually works properly yet as I don't have my scope handy.

RDJunc, 2025-01-27:
What a royal pain!  I also ran into this problem and of course, natually assumed that it was caused by problems with my code or how the compiler was set up.

I rolled the ESP32 Core back to 3.0.4 but left the ESP_DMX library at 4.1.0.  Yes, it does compile and work, at least it did for me.

tracetechnical, 2025-01-27:
Gave it a run up with 4.1.0 of this library myself and confirm this combo works here too.

ANLCRAW, 2025-01-29:
[#181](https://github.com/someweisguy/esp_dmx/issues/181#issuecomment-2619261884) Replace your uart.c from this post. Worked for me!

tracetechnical, 2025-01-29:
> [#181](https://github.com/someweisguy/esp_dmx/issues/181#issuecomment-2619261884) Replace your uart.c from this post. Worked for me!

Rolling back the esp32 BM version is a better idea than patching this library. As soon as you start patching libraries you risk losing them in a future upgrade.

jumpjack, 2025-04-26:
news? Official updates?