someweisguy/esp_dmx #198: unable to compile with 4.1.0 in Arduino for ESP32

thunermay, 2025-04-11:
If I select ESP32 Dev Module, I cannot compile the code; this is the error:
```
/Users/thunermay/Documents/Arduino/libraries/esp_dmx/src/dmx/hal/uart.c: In function 'dmx_uart_init':
/Users/thunermay/Documents/Arduino/libraries/esp_dmx/src/dmx/hal/uart.c:330:51: error: 'uart_signal_conn_t' has no member named 'module'
  330 |   periph_module_enable(uart_periph_signal[dmx_num].module);
      |                                                   ^
/Users/thunermay/Documents/Arduino/libraries/esp_dmx/src/dmx/hal/uart.c:338:52: error: 'uart_signal_conn_t' has no member named 'module'
  338 |     periph_module_reset(uart_periph_signal[dmx_num].module);
      |                                                    ^
/Users/thunermay/Documents/Arduino/libraries/esp_dmx/src/dmx/hal/uart.c: In function 'dmx_uart_deinit':
/Users/thunermay/Documents/Arduino/libraries/esp_dmx/src/dmx/hal/uart.c:384:56: error: 'uart_signal_conn_t' has no member named 'module'
  384 |     periph_module_disable(uart_periph_signal[uart->num].module);
      |                                                        ^
exit status 1

Compilation error: exit status 1
```

This is just for importing (`#include <esp_dmx.h>`)

kkirjala, 2025-04-23:
I have the exact same issue. Arduino IDE version 2.3.6 and ESP32 CP2102 board.

jumpjack, 2025-04-26:
Workaround: https://github.com/someweisguy/esp_dmx/issues/186